LumaTeam/Luma3DS #2072: Volume Override Issues

ItalianNinjaa, 2024-08-23:
**System model:**

[New 2DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[11.17.0-50U SysNAND, 11.17.0-50E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[v13.1.2 stable]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4 )

Splash: ( None )

Splash duration: ( None )

PIN lock: ( None )

New 3DS CPU: ( Default )

Hbmenu autoboot: ( Yes )

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( X )

Enable game patching: ( X )

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: ( X )

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**
So my Volume slider is broken and i use the volume override option in the rosalina menu

When i use the Snes9x emulator, (downloaded on the universal updater) the volume override isn't applied

Same for the apps: GameYob (GB and GBC emulator), VirtuaNES (NES Emulator)

**Steps to reproduce:**

1. Enable the volume override in the rosalina menu and set it to anything

2. Open one of the 3 apps described here (Snes9x, GameYob, VirtuaNES) and load a ROM

3. The volume override isn't working

**Dump file (if applicable):**

ItalianNinjaa, 2024-08-23:
the download isn't working

profi200, 2024-08-23:
These homebrew apps probably use CSND (I2S2) for sound which has a different volume setting than DSP (I2S1) output. Adjusting this setting is trickier because it's used for the camera shutter sound and so it would get overwritten as soon as you use any camera functionality.

ItalianNinjaa, 2024-08-24:
> My account has been stolen and they are sending junk emails or links all
> over the place. I've changed my password. I'm sorry for what you've
> received.
> 1propanol

ok no problem I didn't download the file

commit 77f225ed (2024-10-19): Fix volume override for applets (i2s2/csnd)

This feature repurposes the functionality used for the camera shutter sound.
As such, it interferes with it:
    - shutter volume is set to the override instead of its default 100% value
    - due to implementation details, having the shutter sound effect play will
      make this feature stop working until the volume override is reapplied by
      going baxck to this menu

Closes #2072
commit 9b8a95a1 (2024-10-20): Fix volume override for applets (i2s2/csnd)

This feature repurposes the functionality used for the camera shutter sound.
As such, it interferes with it:
    - shutter volume is set to the override instead of its default 100% value
    - due to implementation details, having the shutter sound effect play will
      make this feature stop working until the volume override is reapplied by
      going back to this menu

Closes #2072
closed 2024-10-19 by TuxSH