LumaTeam/Luma3DS #2074: after playing new super mario bros 2 I can't open homebrew apps

Ghegghi, 2024-08-26:
when I play new super mario bros 2 after I close it, if I try to open a homebrew app it gets stuck on a black screen and i need to force the 3ds to shut down by holding down the power button.

https://github.com/user-attachments/assets/30dc8574-12b5-4abc-ac49-28d5ab8e04c3

TuxSH, 2024-09-15:
After some investigation, NSMB2 seems to set GPU in an unexpected state homebrew doesn't support. A workaround for this is to open and then close any other official application.

Cc @fincs . Transferring to libctru

Searinox, 2024-09-28:
This issue was reported before and is known to happen with both NSMB2 and Castlevania Mirror of Fate. There is an issue filed for this: https://github.com/LumaTeam/Luma3DS/issues/1559
Would be nice if this was fixed on the Luma side as Nintendo will never fix it now.

TuxSH, 2024-10-01:
Ideally this is up to libctru to fix it (and homebrew (especially hbmenu)) to be recompiled with it. I don't have much expertise with 3DS GPU programming so it's up to fincs to investigate it, I think.

TuxSH, 2024-11-20:
Not a luma issue, now fixed by: https://github.com/devkitPro/libctru/pull/554

closed 2024-11-20 by TuxSH