LumaTeam/Luma3DS #2080: An exception occurred (ARM11)

TakumaEN, 2024-09-14:
**System model:**
New 3DS

**SysNAND version:**
11.17.0-50U SysNAND

**Entrypoint:**
Bootes from SD via B9S

**Luma3DS version:**
v13.1.2

**Luma3DS configuration/options:**

Screen brightness: ( 1 )

Splash: ( Off )

PIN lock: ( Off )

New 3DS CPU: ( Off )

Hbmenu autoboot: ( Off )

Enable loading external FIRMs and modules: ( x )

Enable game patching: ( x )

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: ( x )

Show GBA boot screen in patched AGB_FIRM: ( )

**Explanation of the issue:**
My 3DS crashes at random while playing Persona Q2. By the error screen, it says:

> Processor: Arm 11 (Core 0)
> Exception type: data abort
> Fault status: Translation - Section
> Current process: AQ2 (00040000001D7100) 

Not sure if the issue is game related, so I've attempted to see if this happens while playing other games in my system, but so far, not, but that's not to say it couldn't be. This is proving to be quite difficult to troubleshoot, because it really is at random. Sometimes it crashes a few minutes in, and sometimes over the span of 30 minutes or so, it's very unpredictable!

I've left the log in the attachment just in case.

[crash_dump_00000000.zip](https://github.com/user-attachments/files/17002624/crash_dump_00000000.zip)
, it's very unpredictable!

thepokemonsters, 2024-10-30:
im having the same problem. cant boot any games

Aqua-moonn, 2025-02-02:
Also getting this error, and have found no solution anywhere (Google, reddit, gbatemp, and the Nintendo homebrew discord)

Masamune3210, 2025-02-02:
So you are all having issues with Persona Q2 specifically and no other game or app?

TakumaEN, 2025-02-02:
@Masamune3210 yes, as weird as it sounds it is the only game I have downloaded in my system that crashes and returns that error.

urherenow, 2025-02-02:
Then it’s a bad download, or you’re simply missing the seed…

TakumaEN, 2025-02-02:
@urherenow Whoa, I wasn't familiarized with the whole seed thingjust looked a bit into seed, but I wan to try this hypothesis out. Sorry to ask such noob questions, but could you elucidate me on how do I find the seed and where should I place it in the files?

urherenow, 2025-02-02:
Sorry, but no. Piracy is not supported in any way.

TakumaEN, 2025-02-02:
@urherenow oh, it's one of those methods. I see, I see. Well, I'll try looking into it and return with the results. Might take me a while, but thanks, that gives me hope!

closed 2025-02-02 by PabloMK7

Xancletaman, 2025-09-15:
you can fix the problem doing this; get inside fbi, look for the game with the probleam, an choose import seed