LumaTeam/Luma3DS #2084: After activating the option Enable loading external FIRMs and modules on my 2ds, it is giving arm11 (core 1

Chuckiin, 2024-09-24:
Does anyone know if I should have any extra files? I've done everything, replaced the boot.firm manually and my Luma is version 13.1.2
[image]

TuxSH, 2024-09-27:
Hi, what is the content of `/luma/sysmodules` folder?

EderParis, 2024-09-29:
I'm having the same problem, here's the crash dump.

[crash_dump_00000006.dmp](https://github.com/user-attachments/files/17180235/crash_dump_00000006.dmp)

TuxSH, 2024-10-01:
Neither of you have answered the question. In the absence of details, and I think this is not caused by Luma3DS directly, I'll close the issue.

If I have to make a guess, this might be due to an old version of reHID and/or wuumibo that broke upon upgrading to Luma3DS v13.0. If you have these custom sysmodules, please update them to their latest release.

closed 2024-10-01 by TuxSH

EderParis, 2024-10-02:
Sorry I have no idea what I was doing, but I was in fact using and old version of reHID, I just update it and now everything is working fine.
Sorry for taking your time with some unrelated problem.
Thank you so much for helping me.

TuxSH, 2024-10-02:
No problem :)