LumaTeam/Luma3DS #2087: Possible to disable 3DS LEDs?

Darthagnon, 2024-10-06:
I have been using [this unofficial fork of Luma3DS 10.3](https://gbatemp.net/download/luma-10-3-with-night-light-and-quick-switchers.35619/) ([Github](https://github.com/DullPointer/Luma3DS)) because it allows me to disable the LEDs that otherwise blind me when playing in the dark. 

I'm quite keen to update to the latest version of Luma3DS, but I wanted to check - have the changes from that fork been incorporated, or is it now possible to disable the 2/3DS power LEDs via vanilla Luma3DS?

Many thanks!

Darthagnon, 2024-10-06:
So I updated to Luma3DS 13.2.1 from the unofficial fork, and now the system crashes on boot (dump attached)
[crash_dump_00000004.zip](https://github.com/user-attachments/files/17267915/crash_dump_00000004.zip)

Not sure what I did wrong (copy-paste overwrite was all I did to update) or why it's crashing... can anyone please help me?

Darthagnon, 2024-10-06:
Solved: It was ReHID, the older version must be manually removed before updating to Luma3DS 13. ReHID v4.0 can then be installed.

And also, solved: Luma3DS v13 has LED toggles in the Rosalina Menu.

Darthagnon, 2024-10-06:
Having tested, disabling the LEDs is not persistent - every reboot or after every sleep, the LEDs are re-enabled. I need to go into the Rosalina menu again or open [BiPoLAR](https://github.com/Spaqin/bipolar-3ds/) every time. When I was using the aforementioned unofficial Luma3DS fork, my LEDs stayed switched off. 

Any chance you could implement switching LEDs off in the same way, so that it persists? Many thanks!

TuxSH, 2024-10-19:
> So I updated to Luma3DS 13.2.1 from the unofficial fork, and now the system crashes on boot (dump attached)
[crash_dump_00000004.zip](https://github.com/user-attachments/files/17267915/crash_dump_00000004.zip)

Hello :)

Forks are not supported and support will not be provided for those.

> Having tested, disabling the LEDs is not persistent - every reboot or after every sleep, the LEDs are re-enabled. I need to go into the Rosalina menu again

That's normal, MCU resets the LED configuration on sleep entry and reboot.

> Any chance you could implement switching LEDs off in the same way, so that it persists? Many thanks!

I will do so at some point in the future but this is low-priority work.

That-One-Bit, 2025-08-03:
Starlight3ds has this feature available, with much more. It's the most starred fork for the latest version of luma at the time of this comment (13.3.3).

closed 2026-10-27 by TuxSH