LumaTeam/Luma3DS #2089: Rosalina system info incorrectly identifies my screen types as TN. I have can tell I have IPS screens

Ninjajoe95, 2024-10-12:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.17.0-50U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

V13.2.1 stable 

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

Hbmenu autoboot: ( )

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**
Under system info in the Rosalina menu, it identifies my top screen type and bottom screen type as TN. However, all I have to do to tell it's an IPS screen is look at it at a heavy angle, and I can still clearly see the image. Definitely an IPS screen on both screens, not sure why it's misidentifying it.

**Steps to reproduce:**

1. No steps necessary

2.

**Dump file (if applicable):**

TuxSH, 2024-10-13:
The IPS/TN status is sourced from the screen's controller itself. You do not have IPS screens.

>  which also says that my shell is red

RED is the codename for the New 3DS XL

closed 2024-10-13 by TuxSH

Ninjajoe95, 2024-10-13:
https://github.com/user-attachments/assets/6611c5e6-8154-4339-ba64-11e3ce421a23

[image]

I rest my case

TuxSH, 2024-10-14:
The video shows that your two screens are in fact, TN. IPS screens don't whiten from an angle like yours do, if you had an IPS top screen you should still been able to see the battery icon and "Internet" icon clearly with no discoloration.

Evidently (judging from 0:05 in your video), that is not the case.