LumaTeam/Luma3DS #2094: complete freeze when forcing connection, entire system becomes unresponsive

vincentmethinks, 2024-11-05:
**System model:** old 2ds

**SysNAND version (+emu/redNAND version if applicable):** sys 11.17.0-50E 

**Entrypoint (How/what you're using to boot Luma3DS):** boot9strap (i think)

**Luma3DS version:** v13.2.1

**Luma3DS configuration/options:**

Default EmuNAND: ( n/a)

Screen brightness: ( 4)

Splash: ( off)

Splash duration: (n/a )

PIN lock: ( none)

Hbmenu autoboot: ( off)

--

Autoboot EmuNAND: ( n/a)

Enable loading external FIRMs and modules: (on )

Enable game patching: (on)

Show NAND or user string in System Settings: ( on)

Show GBA boot screen in patched AGB_FIRM: (off )

--

**Explanation of the issue:**
i have no idea how or why it happened, but i was having trouble connecting to wifi and tried to use the force connection option. when i did this, the entire system froze and became unresponsive to any input including the sleep switch. it's probably a fault on my part(i have no idea how luma3ds works or how to do anything complicated with it), and i was able to reboot the system by forcing a shutdown and turning it back on, but i haven't found anything online about this issue so i figured id try here. i just want to know what caused the freeze :p 

**Steps to reproduce:**

1. open rosalina menu

2. force connection (to third option if that's important)

3. complete freeze of system

PabloMK7, 2024-11-05:
Hello, this place is intended to keeping track of luma bugs and not really suited for giving assistance. Instead try to ask in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

If your console can't connect to any wifi, either all the networks are incompatible (5GHz) or your 3ds wifi module is faulty.

closed 2024-11-05 by PabloMK7

vincentmethinks, 2024-11-05:
would this not count as a bug? my wifi module isn't faulty to my knowledge because ive connected to this wifi network before without any issues