LumaTeam/Luma3DS #2099: Sound in Mario & Luigi: Dream Team Bros. temporarily stops when Rosalina menu is opened

mariomadproductions, 2024-11-29:
**System model:** New 3DS
**SysNAND version:** 11.17.0-50E
**Entrypoint (How/what you're using to boot Luma3DS):** fastboot3DS
**Luma3DS version:** v13.1.2
**Luma3DS configuration/options:** 
Screen brightness: (4)
Splash: (Off)
PIN lock: (Off)
New 3DS CPU: (Off)
Hbmenu autoboot: (Off)
Enable loading external FIRMs and modules: (Off)
Enable game patching: (On)
Redirect app. syscore threads to core2: (Off)
Show NAND or user string in System Settings: (On)
Show GBA boot screen in patched AGB_FIRM: (Off)

**Explanation of the issue:**
Sound in Mario & Luigi: Dream Team Bros. sometimes stops when Rosalina menu is opened. This can be worked around by opening and closing the home menu after this happens (suspending and unsuspending the game). Happened in one of the pause menus for me. European cart version installed to console. Game's update data v1.1 is also installed. 

**Steps to reproduce:**
See above.

**Dump file (if applicable):**
N/A

cooolgamer, 2024-12-05:
duplicate of #1483

mariomadproductions, 2024-12-05:
> duplicate of #1483

Maybe the cause is the same, but the symptoms aren't.

TuxSH, 2024-12-07:
More or less the same thing and is due to implementation limitation, can't fix.

closed 2024-12-07 by TuxSH