LumaTeam/Luma3DS #2100: SD:/luma/dumps/arm11/crash_dump00000000.dmp help

95brx, 2024-11-30:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.17.0-50U SysNAND, 11.17.0-50E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v13.2.1 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/988ec17ebfce513fc4589f7b12e0d6e3894ae542]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

Hbmenu autoboot: ( )

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file (if applicable):**

95brx, 2024-11-30:
i need help i put the crash dump on the title

PabloMK7, 2024-11-30:
Hello, this is a place to track luma3ds bugs and not suited for assistance. Please ask in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

closed 2024-11-30 by PabloMK7