LumaTeam/Luma3DS #2101: 0xf96047a2 errdisp generic error comes up every time I try to boot one specific 3ds game

amagiisms, 2024-11-30:
**System model:**
 New 3DS

**SysNAND version:**
11.17.0-50U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
12.0

**Luma3DS configuration/options:**

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

Hbmenu autoboot: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X )

Show NAND or user string in System Settings: ( X)

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**
Every time I launch a particular 3DS game (Yo-Kai Watch), no matter if I use the cartridge or the digital version, it gives me the same error. It doesn't do this with any other game, whether it's on cart or digital or emulated. How is it that both the downloaded version and the cartridge have the same exact problem? 

Error looks like this:

Error type: generic
Process ID: 39
Process name: yw1
Process title ID: 0004000000167700
Address: 0x00333ff4
Error code: 0xf96047a2

PabloMK7, 2024-11-30:
Hello, this is a place to track luma3ds bugs and not suited for assistance. Please ask in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

closed 2024-11-30 by PabloMK7

urherenow, 2024-11-30:
I’ll be nice and point this in the right direction, since it’s a recurring issue fr people and it’s been a while since this popped up.

whats the same between the cart and digital? Everything except for the tickets. Your problem is likely that you have a (wrongly) configured region setting for this game. Search your SD card for the title ID, and I think you’ll find a folder with the same name as the id. Correct the file within, or delete it, if it’s not necessary.

PabloMK7, 2024-11-30:
Please, try to move the discussion to a help forum, otherwise we get spammed with messages that are not related to Luma3DS in our mail.