LumaTeam/Luma3DS #2104: Games such as the legend of dark witch 2 doesn't apply layeredfs mods

cooolgamer, 2024-12-05:
Just like Mario Kart 7 when the v1.2 update dropped, some mods (like texts) applied using LayeredFS didn't work because the update used a specific file path.
I noticed that the game "The Legend of Dark Witch 2" also has the same issue, it seems like "ro2:" was somehow detected instead of "rom2:" (I'm not really sure but it looks like that's the case)

The fix would be something like this (in sysmodules/loader/source/patcher.c, this is Starlight3DS, the lines won't really match but this is the same thing for this issue anyways):
[image]
This fixes the issue, I can make a pull request, but it would be better to find something else that doing hacky fix for specific games

cooolgamer, 2024-12-05:
I found the issue, I'm making a pull request about this soon
[image]
- Added a null terminator (and extended temp size for that)
- it was temp + 1, not patternSize + 1

PabloMK7, 2024-12-05:
So, after discussing with @cooolgamer privately, it was realized that the issue is the null byte check before the pattern. Maybe this is too prevent false positives, but it's failing here.