LumaTeam/Luma3DS #2106: ''An Exception Ocurred

TheJohanx, 2024-12-12:
**System model:**

New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.17.0-50U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v13.2.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4)

Splash: (Before )

Splash duration: (1s)

PIN lock: (Off )

New 3DS CPU: (Off )

Hbmenu autoboot: (Off )

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x )

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**

When playing game (Mario & Luigi: Bowser's Inside Story + Bowser Jr. Journey, i randomly get an ''An exception occurred'' error message and i have to shutdown the system

Processor: Arm11 (core 0)
Exception type: data abort
Fault Status: Translation - Section
Current process: Remilio (00040000001D1400)

**Steps to reproduce:**

1.Just playing the game, its apparently random
(Must note that ive been using a 60fps cheat code and the crashes have been more usual, however, i still got these crashes before using the cheat so i dont think its the cheat the cause of it)

**Dump file (if applicable):**
[Crash Dump.zip](https://github.com/user-attachments/files/18112863/Crash.Dump.zip)

vincentmethinks, 2025-01-01:
game or save files are corrupted most likely. if its at the same points, it's probably the game files. at which points does the game crash?

PabloMK7, 2025-01-01:
This is caused by corruption or the cheats being used.

closed 2025-01-01 by PabloMK7