LumaTeam/Luma3DS #2110: Games crashes in last boss' fight in Bravely Second. Anybody can help?

Jofezito, 2025-01-10:
I've been having this problem with Bravely second, in the last battle against Lord Providence, after they start to manipulate the characters and is about to access the main title to exclude your save game, my game crashes. Then, the 2DS powers off and when I try to play again, I'm in my last save area. Anyone knows how to fix it?
[image]
[image]

urherenow, 2025-01-10:
Github is not a help forum. You might want to ask at gbatemp or discord retronx.

Jofezito, 2025-01-10:
> Github is not a help forum. You might want to ask at gbatemp or discord retronx.

Thanks! Will do

closed 2025-01-24 by PabloMK7

Jbubby, 2025-04-03:
hey did u find a solution to this tho? i cant find anything and i seem to be having the same problem!!! but random encounters instead of that boss fight