LumaTeam/Luma3DS #2112: [Audio Issue] Is there any way to adjust the pitch of an N3DS's audio?

DiplacusisAlt, 2025-01-23:
In short, I have an auditory disorder that makes everything sound pitched up by roughly 5% -- enough to make a song in the key of C sound like it's in C#. Is there any way to tweak either the hardware or software of an N3DS in order to adjust the pitch a tad? Any help in this regard would be greatly appreciated!

sakumikimiko, 2025-01-23:
> Is there any way to tweak either the hardware or software of an N3DS in order to adjust the pitch a tad? 

All sounds are linked to the actual game code, or to console hardware, and in order to change the sounds by half a semitone, which is what you are implying, you would need to re-make the sounds for a game and make it a custom mod, or alter the sound used in the hardware of the n3ds. I do not know how to do this, but I don't think there is way you can alter hardware n3ds sounds. You may be able to create/find/buy custom soundpacks for a game however.

PabloMK7, 2025-01-23:
The comment above pretty much sums it up. It _could_ be possible that the audio can be sped up on the DSP level so that the pitch changes too, but then it would cause other audio sync issues. 

Sadly, there is no other way to do it other than a per game basis :(

closed 2025-01-23 by PabloMK7

DiplacusisAlt, 2025-01-23:
Agh, unfortunate. Thank you both for your answers!

profi200, 2025-01-24:
Adding to the above the CODEC chip which turns the digital audio into analogue has various filters and settings. I don't know if the IIR filters are capable of doing such processing but it's worth mentioning them. If the filters can't there is a small chance you could archive something by tweaking the clock dividers to change the sample rates on the CODEC side.