LumaTeam/Luma3DS #2114: An exception occurred

ilove2ds, 2025-02-02:
**System model:**

Old 2ds

**SysNAND version (+emu/redNAND version if applicable):**

i dont know my SysNAND version of the top of my head and i cannot boot into the 2ds menu

Version: Luma3DS v13.3

[e.g. v13.2.1 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/988ec17ebfce513fc4589f7b12e0d6e3894ae542]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4 )

Splash: ( off)

Splash duration: ( )

PIN lock: ( off)

New 3DS CPU: ( )

Hbmenu autoboot: (off )

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( x)

Enable game patching: (x )

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**
i can boot into gm9 and luma3ds however, when i try to boot normally i have the error:

An exception occurred
Processor: Arm11(core 1)
Exception type: prefetch abort (svcBreak)
Fault status: Debug event
Current process: ns (0004013000008002)
it happened when i was playing a .nds rom of GTA through the NDSforwarder from universal updater, i clicked home and after moment of black screen i received that error.
I have tried updating luma3ds and i haven't updated anything else before or after.

**Dump file (if applicable):**

the time is not correct on my but these dumps were made yesterday and today

[crash_dump_00000000.zip](https://github.com/user-attachments/files/18632653/crash_dump_00000000.zip)
[crash_dump_00000001.zip](https://github.com/user-attachments/files/18632657/crash_dump_00000001.zip)
[crash_dump_00000002.zip](https://github.com/user-attachments/files/18632656/crash_dump_00000002.zip)
[crash_dump_00000003.zip](https://github.com/user-attachments/files/18632654/crash_dump_00000003.zip)
[crash_dump_00000004.zip](https://github.com/user-attachments/files/18632655/crash_dump_00000004.zip)

ilove2ds, 2025-02-02:
i forgot to mention that it can boot without an sd card.

PabloMK7, 2025-02-02:
Hello, this issue is probably caused by a corrupted SD. For further help please ask in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

closed 2025-02-02 by PabloMK7