LumaTeam/Luma3DS #2115: [Feature Request] Profiles for screen filters

mlgmaster666, 2025-02-04:
Title basically says everything, i just think it would be nice to have different config profiles for different apps/games, especially after the introduction of the IPS fix option.

PabloMK7, 2025-02-04:
Is there really a use case for using different filter parameters for different games? If I were to setup a screen filter I would only do it once and not switch every game.

mlgmaster666, 2025-02-04:
mostly bc of the ips fix option, some games look really good with it on, especially emulators, but some are already trying to  compensate for the 3dss original colors, and become a bit saturated, its also know to potentially make the black crush even worse on some titles.  

And since mine only has one IPS, i need to manually configure the other screen to match, so it becomes a bit time consuming to turn it on/off.

crvptid, 2025-04-24:
If different profiles attached to different games is too niche, might I suggest being able to save presets for screen adjustments? Maybe that's what's being suggested already, but if so I second it for a different use case. I'd really love to have an option to quick-change to my 'night mode' redshift & screen settings, since it can take a while to switch it back and forth. (Bonus if LED on/off status can be saved with it too, lol)