LumaTeam/Luma3DS #2116: Please help!

damiandoky37, 2025-02-22:
Whenever i stay on the 3ds homescreen for like 10 seconds it crashes and says that "boss" chashed the system. I tried deleting some unused games but that didnt work. dmp:

[crash_dump_00000000.zip](https://github.com/user-attachments/files/18918221/crash_dump_00000000.zip)

PabloMK7, 2025-02-22:
Hello, this place is for tracking Luma3DS bugs and not for assistance, please ask in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

In any case this looks like a corrupted SD. Copy everything to your PC, format the SD card, then copy everything back.

closed 2025-02-22 by PabloMK7

damiandoky37, 2025-02-23:
> Hello, this place is for tracking Luma3DS bugs and not for assistance, please ask in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh
>
> In any case this [...]

will it brick if i do this and have you done it yourself?

randompersononinternet69, 2025-02-25:
> will it brick if i do this and have you done it yourself?

it shouldn't brick

damiandoky37, 2025-03-03:
eh i just went to the discord and they helped me. turns out it was just netpass being faulty