LumaTeam/Luma3DS #2119: An exception occurred. ARM11 (core 0) - data abort - Translation ~ Page - momiji (00040000001B5100) & An exception occurred. ARM11 (core 0) - data abort - Alignment - momiji (00040000001B5100)

ZaraEspeon, 2025-03-02:
**System model:**

[Old 3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[11.17.0-50U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[v13.3 configuration]

**Luma3DS configuration/options:**

Screen brightness: 4(x) 3(  ) 2( ) 1(  )

Splash: Off(x) Before (0) After(  ) payloads 

PIN lock: Off(x) 4(x) 6(  ) 8(  ) digits

Hbmenu autoboot: Off(x) 3DS(  ) DSi ( )

--

(x) Enable loading external FIRMs and modules

(x) Enable game patching

(x) Show NAND or user string in System Settings

(x) Show GBA boot screen in patched AGB_FIRM

--

**Explanation of the issue:**

I have been using Luma for a while now for my Pokemon 3DS from hacks. I recently pached Ultra Moon with Penumbra Moon. Everything was working fine. I wanted to add a Pokemon from PKheX to my game, one I would not have access to until much later, so I installed JKSM and FBI soon afterward, creating a cias folder for it. After I added the Pokemon from PKheX and overwrote the save "main" of the game, whenever I start the game, it will randomly give one of these two errors at different points in the game with no clear indication of the trigger (e.g. walking, battling, during text, etc.) 

**Steps to reproduce:**

1. Start console with checked HBmenu options listed above.

2. Boot game

3.  Play normally; error will randomly occur at different times with no apparent trigger, but two different Fault Statuses

**Dump file (if applicable):**

I've attached all the dump files. The most recent one is crash_dump_00000005.dmp

Please advise how I can resolve this consistent crash of my game.

[Dump Files.zip](https://github.com/user-attachments/files/19042240/Dump.Files.zip)

PabloMK7, 2025-03-02:
Hello, the bugs you have mentioned seem to be related to the tools you have used. Luma only shows the error screen, but is not responsible for the crashes. Please try to reach the developers of the respective tools you have used.

closed 2025-03-02 by PabloMK7

ZaraEspeon, 2025-03-05:
I am still having issues. I tried to reach out on JKSM, but have not heard a response yet. It seems that whenever I get to a point where I have to type information in (e.g. nickname my Pokemon) it freezes with the Translation error. I have included the most recent dump files. Is there anyone else I can contact about this, please? 

[More Recent Dumps.zip](https://github.com/user-attachments/files/19045346/More.Recent.Dumps.zip)

[Most Recent Dump.zip](https://github.com/user-attachments/files/19043892/Most.Recent.Dump.zip)

[Dump Files 3.zip](https://github.com/user-attachments/files/19043999/Dump.Files.3.zip)