LumaTeam/Luma3DS #2121: Circle pad pro connection lost [MH4U]

Aepoh, 2025-03-03:
**System model:**

[New 3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[11.16.0-49U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[v13.3]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: (*)

Hbmenu autoboot: ( )

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: (*)

Enable game patching: (*)

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: (*)

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**
MH4U loses connection to the c stick (circle pad pro) when the system is closed and enters sleep mode.

https://gbatemp.net/threads/help-c-stick-disconnect-in-mh4u-on-n3ds-xl.593698/

**Steps to reproduce:**

1. Launch MH4U. Circle pad pro should initially connect.

2. Enter sleep mode and then reopen the system. There's a good chance the circle pad pro will disconnect and the C stick will no longer work.

**Dump file (if applicable):**

Aepoh, 2025-03-11:
Upon further testing it seems the Load External Firms and Modules option is the culprit. The bug does not occur when that is turned off.

Aepoh, 2025-03-18:
It's the Enable Firms and Modules option that causes the issue.

Aepoh, 2025-04-07:
Hello?
Also I experience this bug on a downloaded version of MH4U. I do not know if occurs on physical cartridge.

TuxSH, 2025-04-08:
This not a Luma3DS issue as it has nothing to do with circle pad pro unless "input redirection" is enabled in rosalina menu, which I don't think you have enabled.

> Upon further testing it seems the Load External Firms and Modules option is the culprit. The bug does not occur when that is turned off.

This is a reHID issue then. Check your reHID version and configuration.

Aepoh, 2025-04-09:
That doesn't make sense as for the longest time I experienced this bug without having a ReHID profile set for MH4U. When I did make a ReHID profile for MH4U the bug still occurred. When I deleted that .json file for MH4U, the bug still occurs. I deleted the title folders for ReHID and reinstalled it and updated it, and the bug still occurs.

The bug occurs on both digital and physical versions of MH4U. The bug only occurs when "Load External FIRMSs and Modules" is turned on in Luma boot settings. The bug does not occur when this is turned off. From this information there is no reason to believe that ReHID is the culprit, and there is EVERY reason to believe that Luma is the culprit. This has been a problem for years. Did you check that gbatemp forum link I posted? I am not he only one experiencing this issue.

Aepoh, 2025-04-09:
I have done a more thorough removal of ReHID following instructions from a comment on this reddit thread: https://www.reddit.com/r/3dspiracy/comments/1auwwcc/problem_with_rehid/?rdt=33709
I will report back if the bug continues to occur or not. If it does, it most definitely not ReHID's fault and very very likely Luma's fault.

Aepoh, 2025-04-09:
The bug did not reoccur but was replaced by another issue. When "Load External firms and modules" was on, the ZL and ZR buttons would not work in games even when the C stick would work. ZL and ZR worked as normal in the home menu. I tested this with MH4U and Terraria.

After doing a more thorough removal of ReHID, the bug does not occur even with the external firms/modules settings turned on, and the ZL and ZR buttons work as expected.