LumaTeam/Luma3DS #2126: my button L is broken

deidv, 2025-03-22:
My console has a broken L button, because of this I can't use many of the options in Rosalina's menu, such as screenshots or using non-physical amiibos. Is there any way I can launch Rosalina's menu without having to use the L button? Is there any way to change the button configuration to open it? Or am I screwed?

impeeza, 2025-03-22:
You can change the key combo on /luma/config.ini on the rosalina_menu_combo option

deidv, 2025-03-22:
> You can change the key combo on /luma/config.ini on the rosalina_menu_combo option
thank u, u saved my life <3

deidv, 2025-03-22:
> You can change the key combo on /luma/config.ini on the rosalina_menu_combo option

but wait, how can i do that?, im kinda new in this world

closed 2025-03-22 by deidv

impeeza, 2025-03-23:
JUST OPEN THE FILE SDMC:/LUMA/CONFIG.INI ON YOUR SD CARD WITH A GOOD TEXT EDITOR:

[image]

AND CHANGE THIS LINE, YOU CAN USE THE NEXT VALUES FOR EACH KEY:

"A", "B", "Select", "Start", "Right", "Left", "Up", "Down", "R", "L", "X", "Y",
"ZL", "ZR",
"Touch",
"CStick Right", "CStick Left", "CStick Up", "CStick Down",
"CPad Right", "CPad Left", "CPad Up", "CPad Down"

SAVE THE FILE AND RESTART THE CONSOLE.

THIS SITE IS NOT FOR SUPPORT, IS ONLY TO REPORT BUGS OR PROBLEMS.

PLEASE IF YOU HAVE FURTHER QUESTIONS GO TO DISCORD OF GBATEMP.NET THERE YOU WILL RECEIVE HELP FROM THE SCENE.

Masamune3210, 2025-03-24:
why are you screaming lol

impeeza, 2025-03-24:
MY VOLUME KNOW IS STUCKED AT 100% 😂🤣