LumaTeam/Luma3DS #2127: Mixed WPA2/3 WiFi security appears to be incompatible with the system

Coolumin, 2025-03-24:
**System model:**

Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.17.0-50U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Fastboot3DS

**Luma3DS version:**

v13.3

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4 )

Splash: ( Off )

Splash duration: ( )

PIN lock: ( Off )

New 3DS CPU: ( )

Hbmenu autoboot: ( Off )

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( X )

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: ( X )

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**

If you have a WiFi access point configured with mixed WPA2/3 encryption and try to connect a 3DS to it, it will display a 003-0299 error code when trying to connect to the access point, even though the console's WiFi is still enabled.

It will only allow you to connect to the access point if you lower its security settings to either exclusively WPA2, WEP, WPA, or mixed WPA/WPA2. 

This might be a hardware limitation, and/or out of scope for Luma, I'm not sure. This has been resolved in things like switchroot Android though, so my assumption is that it could potentically be fixed in software. 

**Steps to reproduce:**

1. Scan for the WiFi access point with WPA2/3 mixed security with the 3DS console (or manually enter the access point settings)

2. Try to run the connection test after adding the access point settings

**Dump file (if applicable):**

urherenow, 2025-03-24:
Switchroot is for the Switch, not the 3ds. 3ds is not capable of wpa3 or 5ghz.

Coolumin, 2025-03-27:
> Switchroot is for the Switch, not the 3ds. 3ds is not capable of wpa3 or 5ghz.

I'm not talking about exclusively WPA3 security, I'm talking about WPA2/3 mixed security. Even the PSP supports that if you use a WPA2 plugin! If I recall correctly, the Wii/U should too.

urherenow, 2025-03-27:
3DS isn't a PSP or WiiU, either. Not sure why you insist on bringing up other systems. Also not sure what posessed you to reply to the issue after it was closed. Best you can do with the 3DS is to set your 2.4ghz radio to wpa2, and set your 5ghz to wpa2/3.