LumaTeam/Luma3DS #2128: Luma3DS BMP screenshots' filename dates seem affected by the Year 2036 Network Time Protocol rollover bug

DandelionSprout, 2025-03-24:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.17.0-50E

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S/Sighax, from what I can tell.

**Luma3DS version:**

v13.3.0

**Luma3DS configuration/options:**

Default EmuNAND: SysNAND from what I can tell. While my 3DS does have "EmuNAND 1" and "EmuNAND X", I couldn't spot any indications that either of the 2 EmuNANDs were in use.

Screen brightness: 4

Splash: Off

PIN lock: None

Hbmenu autoboot: Off

--

Autoboot EmuNAND: I seem to be using SysNAND, so no.

Enable loading external FIRMs and modules: No.

Enable game patching: Not from what I can tell.

Show NAND or user string in System Settings: Yes.

Show GBA boot screen in patched AGB_FIRM: No.

--

**Explanation of the issue:**

The screenshots' default filenames are affected by a time rollover bug, very likely the NTP Year 2036. It noticeably resembles the Unix time bug (a.k.a. Year 2038 Problem), but is not the exact same thing.

When it does roll over, the filenames' timestamps deduct approximately 136 years, so that a screenshot taken with the system clock set to 2048 is shown as "1912-(…).bmp".

The date and clock timestamps also become wrong, so that screenshots taken with the system date set to 24 March 2039 23:56 (UTC+1) is timestamped as "1903-02-16_17-28-21.(...).bmp"

Screenshotting the screens and viewing the image files remain functional nevertheless.

**Steps to reproduce:**

1. Set the Nintendo 3DS system date to any time between 8th of February 2036 and late 2050.

2. Take a screenshot with Luma3DS.

3. Use a PC SD-to-USB adapter to see the screenshot in File Explorer (In my case Windows 11's File Explorer).

4. See the mismatched times in the automatic filenames that were generated during the screenshotting.

**Dump file (if applicable):**

Masamune3210, 2025-03-25:
If in 11 years, the only thing that has broken down is the automatic timestamping of screenshots, I think people will be overjoyed

TuxSH, 2025-04-08:
Will have a loog, thanks for reporting

closed 2026-10-27 by TuxSH