LumaTeam/Luma3DS #2135: Download Play region-free asks client to update, even though it's updated

Aquaticsanti, 2025-04-27:
**System model:**

New 3DS XL, Old 2DS

**SysNAND version (+emu/redNAND version if applicable):**

11.17.0-50U (New 3DS XL), 11.17.0-50E (Old 2DS)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap (New 3DS XL, Old 2DS)

**Luma3DS version:**

v13.2.2 stable (Both)

**Luma3DS configuration/options:**

Screen brightness: (4, Both)

Splash: (Before, New 3DS XL) (Off, Old 2DS)

Splash duration: (Default, Both)

PIN lock: (Off, Both)

New 3DS CPU: (Off, New 3DS XL)

Hbmenu autoboot: (Off, Both)

--

Enable loading external FIRMs and modules: (Yes, Both)

Enable game patching: (Yes, Both)

Redirect app. syscore threads to core2: (No, New 3DS XL)

Show NAND or user string in System Settings: (Yes, Both)

Show GBA boot screen in patched AGB_FIRM: (No, Both)

--

**Explanation of the issue:**
If you try to use Download Play region-free between an EUR console and a USA one (Both ways), both consoles see each other, but the download play app tells the client to update, even if their console is updated. I've tried this on Mario Party Island Tour and Mario Kart 7.

**Steps to reproduce:**

1. Start a download play game in the host console, and make it search for other players

2. In the client console, open the Download Play app and select "Nintendo 3DS".

3. After the client is done searching, connect to the host console

The client will be prompted to update, even if they're up to date.

urherenow, 2025-04-27:
This is not a bug. Download Play was never intended to work across regions, and that’s Nintendo’s doing. Please read this thread and close the issue: https://gbatemp.net/threads/region-locked-download-play.529386/

Aquaticsanti, 2025-04-28:
Oh, I thought this was a thing, since it's mentioned in both [v7.0](https://github.com/LumaTeam/Luma3DS/releases/tag/v7.0) and [V10.2.1](https://github.com/LumaTeam/Luma3DS/releases/tag/v10.2.1)

urherenow, 2025-04-28:
I stand corrected. Should have finished reading that thread myself, as the OP didn’t fix the issue with the uela thing. One of the devs should answer this one.