LumaTeam/Luma3DS #2136: An exception occurred

smolseal, 2025-05-06:
**System model:**
New 3DS LL

**SysNAND:**
11.17.0-50U SysNAND

**Entrypoint:**
Boot9Strap

**Luma3DS version:**
v13.0.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Before)

PIN lock: (Off)

New 3DS CPU: (Off)

Hbmenu autoboot: (Off)

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**
Arm11 Exception Error screen shows after launching Animal Crossing: Happy Home Designer (cartridge).
Both the console and the cartridge are Japan Zone.
Other cartridges, also Japan, work fine on this console.

Checked on FBI, Animal Crossing: Happy Home Designer size is 0.00 B. (Checked other cartridges just in case, Story of Seasons,
the size is 117.98 MiB)

**Steps to reproduce:**

1. Launch the game.

2. "An exception occurred" under Arm 11 (core 0).

**Dump file (if applicable):**

[arm11.zip](https://github.com/user-attachments/files/20063129/arm11.zip)

P.S. English isn't my First language, so forgive me if I confuse you.
And I'm a total noob when it comes to modding 3DS. Please explain it like I'm five.
Could you please let me know if the problem roots from the cartridge or my 3DS?
I bought this cartridge secondhand like always and need to know if I have to return it or not.

smolseal, 2025-05-06:
[image]
[image]
[image]

Photos for what it's worth.

PabloMK7, 2025-05-06:
Hello, this place is for tracking Luma3DS bugs and not really suited for assistance. Please ask for support in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh