LumaTeam/Luma3DS #2137: Color filter glitch occurs when using DSi Autoboot with Power Saving mode enabled

Soto-Jnthan, 2025-05-07:
**System model:**

New 3DS LL

**SysNAND version (+emu/redNAND version if applicable):**

11.17.0-50E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Default entrypoint

**Luma3DS version:**

v13.3.2 Stable

**Luma3DS configuration/options:**

[meta]
config_version_major = 3
config_version_minor = 13

[boot]
autoboot_emunand = 0
enable_external_firm_and_modules = 1
enable_game_patching = 1
app_syscore_threads_on_core_2 = 1
show_system_settings_string = 0
show_gba_boot_screen = 1
default_emunand_number = 1
brightness_level = 1
splash_position = off
splash_duration_ms = 3000
pin_lock_num_digits = 0
app_launch_new_3ds_cpu = l2
autoboot_mode = dsi

[rosalina]
hbldr_3dsx_titleid = 000400000d921e00
rosalina_menu_combo = L+Down+Select
plugin_loader_enabled = 0
ntp_tz_offset_min = 0

[screen_filters]
screen_filters_top_cct = 6500
screen_filters_bot_cct = 6500
screen_filters_top_color_curve_adj = 0
screen_filters_bot_color_curve_adj = 0
screen_filters_top_gamma = 1
screen_filters_bot_gamma = 1
screen_filters_top_contrast = 1
screen_filters_bot_contrast = 1
screen_filters_top_brightness = 0
screen_filters_bot_brightness = 0
screen_filters_top_invert = 0
screen_filters_bot_invert = 0

[autoboot]
autoboot_dsi_titleid = 0003000453524c41
autoboot_3ds_app_mem_type = 0

[misc]
force_audio_output = off
volume_slider_override = -1
use_dev_unitinfo = 0
enable_dsi_external_filter = 0
disable_arm11_exception_handlers = 0
enable_safe_firm_rosalina = 0

**Explanation of the issue:**

When using the Luma's DSi Autoboot feature and the 3DS is set to be in Power-Saving Mode from boot, both screens get unwanted color filters as soon the DSi title is launched (that oscillate from pinkish, blue and yellow depending on the content on-screen)

**Steps to reproduce:**

1. Turn on the Power-Saving mode via the 3DS Home Menu Settings 

2. Set up the DSi Autoboot to launch a DSiWare title using the config.ini file

3. Reboot the console

4. Observe the modified colors on-sceen once the title has launched

TuxSH, 2025-05-07:
A few questions:
- IPS or TN screen(s)?
- what happens when you boot the DSi application with the native resolution option (press start or select when booting a ds(i) game -- I'm not sure it will work out of the box; if it doesn't: start a DS(i) game in native res, shut down, _then_ enable dsi autoboot).

Soto-Jnthan, 2025-05-07:
- I have top and bottom TN screens.
- When I boot the DSi title with the native resolution from the HOME Menu, I have no issues at all. When I enter native resolution using dsi autoboot, the issue still remains (I had set dsi autoboot, rebooted the console, waited a second to press enter to avoid opening the boot chainloader and it launched with the native resolution straight from boot).

Another thing I noticed is that I do not even need to have set Power-Saving Mode before reboot, as I still get the same issue by pressing the HOME button and switching it on with Start + D-Pad left and only if I had entered with autoboot.

TuxSH, 2025-05-07:
Erm, sorry if I wasn't precise enough: with the native resolution mode, does the issue (color changing) cover the whole screen or just the displayed part?

Soto-Jnthan, 2025-05-08:
With native resolution I can only see the colors changing over the displayed part and when I press the HOME menu to exit the title, the confirmation message goes back to normal resolution mode and it is affected as before.

Yes, this happens even when I don't bring up the luma menu/chainloader/splash

TuxSH, 2025-05-08:
The issue is caused by improper local-dimming/backlight power saving  LUT passed around, it seems.

I think the bug comes from this line, specifically: https://github.com/LumaTeam/Luma3DS/blob/master/arm9/source/deliver_arg.c#L74 (should be > 248).

Could you please check if:  [boot.zip](https://github.com/user-attachments/files/20107150/boot.zip) fixes your issue? Thanks

Soto-Jnthan, 2025-05-08:
I have got the exact same behavior with both the default and the native resolution mode. Just to confirm, the boot.zip only contained boot.firm, which I have moved to the SD card's root directory.

TuxSH, 2025-05-08:
What about this ([boot.zip](https://github.com/user-attachments/files/20108918/boot.zip))? Though I doubt it's going to change anything.

Soto-Jnthan, 2025-05-08:
I tried them all and I was not able to spot a difference in behavior.

It seems to follow a pattern in which the top screen goes: blue -> red -> reddish tint and bottom screen goes: pink -> yellow ->  blueish tint. And when I press the HOME button it flickers between all those colors before settling at yellowish. https://imgur.com/a/miRydmX

TuxSH, 2025-05-12:
Thanks for having reported the issue. It turns out this was a nasty hardware bug/a wrong assumption in Nintendo's code.

As per my commit message:

> Even though (when running TWL/AGB FIRM) the SoC is in O3DS mode, and the GPU also is,
as well as most other components behaving as such (external RAM, L2C not usable, etc.),
this is NOT the case for the LCD and adaptive backlight logic which retains FULL N3DS
functionality, including a feature where the window is blended with a given color depending
on the overall relative luminance of that window.
>However, Nintendo's own code mistakenly assumes the opposite, and clearly so ("if GPU in N3DS mode"
checks, not passing N3DS extra adaptive backlight (ABL) to TWL/AGB_FIRM). This has implications:
>- Powersaving (ABL) settings in TWL/AGB_FIRM is inconsistent with *both* O3DS (because the new RGB blend
LUT has been set to its current value by NATIVE_FIRM) and N3DS (because "pwn_cnt" and "inertia" are missing
their N3DS-only bits)
>- "rave party" when booting into TWL/AGB_FIRM or O3DS NATIVE_FIRM without these regs (well, the LUT)
initialized. Easiest way to do so is by leveraging the "DSi autooboot" feature Luma provides. It is worth
noting at least the LUT survives hardware reboots (if Nintendo were using DSi software that was using
TLNC-based reboots, they wouldn't have noticed).

commit a9652804 (2025-05-12): bootloader: zerofill N3DS adaptive backlight when autobooting DSi title

Closes #2137.

Even though (when running TWL/AGB FIRM) the SoC is in O3DS mode, and the GPU also is,
as well as most other components behaving as such (external RAM, L2C not usable, etc.),
this is NOT the case for the LCD and adaptive backlight logic which retains FULL N3DS
functionality, including a feature where the window is blended with a given color depending
on the overall relative luminance of that window.

However, Nintendo's own code mistakenly assumes the opposite, and clearly so ("if GPU in N3DS mode"
checks, n [... cut]
commit 0a9f25a6 (2025-05-12): bootloader: zerofill N3DS ABL regs when autobooting DSi title

Closes #2137.

Even though (when running TWL/AGB FIRM) the SoC is in O3DS mode, and the GPU also is,
as well as most other components behaving as such (external RAM, L2C not usable, etc.),
this is NOT the case for the LCD and adaptive backlight logic which retains FULL N3DS
functionality, including a feature where the window is blended with a given color depending
on the overall relative luminance of that window.

However, Nintendo's own code mistakenly assumes the opposite, and clearly so ("if GPU in N3DS mode"
checks, not passing [... cut]
commit 59543da2 (2025-05-12): bootloader: zerofill N3DS ABL regs when autobooting DSi title

Closes #2137.

Even though (when running TWL/AGB FIRM) the SoC is in O3DS mode, and the GPU also is,
as well as most other components behaving as such (external RAM, L2C not usable, etc.),
this is NOT the case for the LCD and adaptive backlight logic which retains FULL N3DS
functionality, including a feature where the window is blended with a given color depending
on the overall relative luminance of that window.

However, Nintendo's own code mistakenly assumes the opposite, and clearly so ("if GPU in N3DS mode"
checks, not passing [... cut]
closed 2025-05-12 by TuxSH