LumaTeam/Luma3DS #2138: Arm11 Exception when launching homebrew which uses romfs files.

sn76496, 2025-05-08:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

Sys 11.17.0-50U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v13.2.2 stable 

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

Hbmenu autoboot: ( )

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Redirect app. syscore threads to core2: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

--

"An exception ocurred" screen whenever I load a homebrew that loads a file stored in romfs folder.

I have built the 3ds-examples using the latest devkitARM toolchain using WSL2 on Windows 10.
The first time I installed the CFW (following all the steps of the 3ds.hacksguide) launching i.e. the romfs.3dsx example app worked fine.   But after using ftpd for a while to send other .3dsx files the problem started to occur.

And now it happens all the time.  I have tried four different sd cards and the issue repeats.

Other examples work fine.

I haven not changed any setting from luma config.

**Steps to reproduce:**

1. Build the romfs example app.

2. Copy to the sd card and launch it using Homebrew Launcher.

**Dump file (if applicable):**

[crash_dump_00000000.zip](https://github.com/user-attachments/files/20112833/crash_dump_00000000.zip)

sn76496, 2025-05-10:
I did some research and got this from gdb:

```
(gdb) continue
Continuing.
[New Thread 45.268]
[New Thread 45.272]

Thread 1 received signal SIGSEGV, Segmentation fault.
navigateToDir (mount=0x0, ppDir=0x8007f34, ppDir@entry=0x8007f2c, pPath=0x8007f2c, pPath@entry=0x8007f24,
    isDir=isDir@entry=false)
    at /home/davem/projects/devkitpro/pacman-packages/libctru/src/libctru-2.4.1/libctru/source/romfs_dev.c:413
warning: 413    /home/davem/projects/devkitpro/pacman-packages/libctru/src/libctru-2.4.1/libctru/source/romfs_dev.c: No such file or directory
(gdb) backtrace
#0  navigateToDir (mount=0x0, ppDir=0x8007f34, ppDir@entry=0x8007f2c, pPath=0x8007f2c, pPath@entry=0x8007f24,
    isDir=isDir@entry=false)
    at /home/davem/projects/devkitpro/pacman-packages/libctru/src/libctru-2.4.1/libctru/source/romfs_dev.c:413
#1  0x001029c4 in romfs_open (r=0x11c488 <_impure_data>, fileStruct=0x800c34c, path=<optimized out>, flags=0,
    mode=438)
    at /home/davem/projects/devkitpro/pacman-packages/libctru/src/libctru-2.4.1/libctru/source/romfs_dev.c:544
#2  0x00119620 in _open_r ()
#3  0x0010e2f4 in _fopen_r ()
#4  0x00100a3c in printfile (path=path@entry=0x11a058 "romfs:/folder/file.txt")
    at /home/stalker/3ds-dev/devkitpro/examples/romfs/source/main.c:8
#5  0x00100b40 in main () at /home/stalker/3ds-dev/devkitpro/examples/romfs/source/main.c:39
(gdb)
```

I have tested other homebrew files downloaded from the Universal Updater and they all run fine.  The files that crash are the ones I build myself.  And only the ones that loads any file from romfs.

So I am wondering, could the problem be related to the devkitARM toolchain?  I will try to do more debugging to the libctru library code.

urherenow, 2025-05-10:
Wsl… why? Is says it can’t find one of your files, and it might be a problem with how many characters are in that path (or not… I’m not going to try counting on my phone). devkitpro on linux usually lives in /opt/devkitpro.

Msys2 might work better for you. Wsl has always been full of quirks (especially with paths). Msys2, or just use a vm.

sn76496, 2025-05-12:
Thanks @urherenow.  I followed your advice and setup a VM using Hyper-V.  

And ... it works now!  I don't know what is wrong in WSL but the romfs example works fine built on the Debian 12 VM.

Will do more tests now.

WinterMute, 2025-07-04:
> Thanks [@urherenow](https://github.com/urherenow). I followed your advice and setup a VM using Hyper-V.
> 
> And ... it works now! I don't know what is wrong in WSL but the romfs example works fine built on the Debian 12 VM.
> 
> Will do more tests now.

This was poor advice and there is nothing wrong with WSL. The most likely explanation is the lag between the release of devkitARM (May 5th) and the release of libctru (May 11th) and a newlib ABI change which meant that libctru needed recompiled with the latest devkitARM. You can simply run dkp-pacman -Syu on WSL and everything should work as expected.

This isn't a Luma3DS issue and would have been better reported on the [devkitPro forums](https://devkitpro.org).