LumaTeam/Luma3DS #2151: [Feature Request] Button remapping

spiritedsnowcat, 2025-06-15:
Hey there,

I have a broken 3DS that I've poured days into trying to fix. The home button is broken, and the latch is impossible to find online. It's apparently a very common break. It'd be really nice if Luma provided a way to remap buttons. For example, Start + Select remaps to Home. That way, broken buttons can be gotten around.

Thanks for your time,
Spirited

spiritedsnowcat, 2025-06-16:
> > The home button is broken, and the latch is impossible to find online. 
> 
> https://de.aliexpress.com/item/1005006660110805.html

I appreciate it, but the issue is the ribbon latch, not the ribbon itself. Thanks for the effort.

Skwashlepro, 2025-07-03:
this would be cool for people wanting to save some money on motherboards and daughter boards. I have a new3ds that has a broken d pad down and i don't wanna pay 20bucks for a 0.50cts part

Skwashlepro, 2025-07-03:
OP, you could use ReHid

PabloMK7, 2025-07-03:
Hello, please use rehid.

closed 2025-07-03 by PabloMK7