LumaTeam/Luma3DS #2155: Trying to reinstall luma3ds but still getting error.

NoctBun, 2025-06-30:
I got my 2ds XL from someone else modded, but I lost the SD card it came with so it's been practically unplayable. I figured to try and fix it by reinstalling the software, got the latest released added to the SD and only got two changes; that it starts the luma3ds but still getting an error, but different. 

My screen got cracked somehow so I can't really read the code it's showing well: 

R0 E0E01BF5 
R2 00000000
R4 00001000
R6 FFFFFFFE
R8 00000000
R10 00000000
R12 0000 --- ( this is where it's screwed up where i can't read it ) 
LR --- ( can't read it ) 
CPSR 

Press A to save the crash dump? 
Press any other button to shut down 

Stack Dump: 

0FFFFFD8: 00 10 00 00 A4 0D 10 00
0FFFFFE0: 00 00 00 00 00 00 00 00
0FFFFFE8: FF FF FF FF 88 0D 10 00 
0FFFFF0: 00 00 00 00 00 00 00 00
0FFFFFF8: 00 00 00 00 04 0D 10 00

And this is how the file is so far, I've never done this so I might've done it wrong: 

[image]

NoctBun, 2025-07-01:
....

Masamune3210, 2025-07-03:
You waited for all of a day and a half. People are not paid to do this, its entirely voluntary. Climb down off the high horse lol.

Considering the folder is named after the archive, I'm going to guess you accidentally misunderstood the install instructions and there's one too many folders. Delete the luma folder, grab the archive again, open it and drag the contents onto the root of the drive

closed 2025-07-03 by PabloMK7