LumaTeam/Luma3DS #2164: is it possible to turn off stereo render using Luma3ds?

semenogka, 2025-07-27:
in some games and applications such as Hey pikmin! and FBI the ability to turn on 3D is disabled. There are no artifacts in them on my O3DS. But in games and applications where it is possible to turn on 3D, I get a strange vertical line of black across the entire height of the screen, black in one pixel. If you turn on 3D and close your left eye, this line disappears. Is it possible to use Luma3ds to make the render in all games the same as in games/applications where it is not possible to turn on 3D? In addition, I will say that this line is there even if you do not turn on 3D. I apologize in advance if this is the wrong place for such a question.

Masamune3210, 2025-07-28:
Wouldn't it be easier to just....leave the slider all the way down on 2D?

semenogka, 2025-07-29:
> Wouldn't it be easier to just....leave the slider all the way down on 2D?

For example, there is a game called Mario Kart. This game has the ability to turn on 3D. I don't turn on 3D and this vertical line is still there (that is, my 3D slider is in the lowest position). And in games like BOXBOY or Hey Pikmin, where there is no option to turn on 3D, this line is not there. I don't turn on the stereoscopic effect at all, but in games where you can turn it on there are lines, and in games where you can't, there are no lines.  I hope everything is clear now.

profi200, 2025-07-30:
Go to system settings and block 3D via parental controls.

ihaveamac, 2025-08-01:
The 3D light is on, indicating that 3D is not really fully disabled. If Parental Controls is enabled and you specifically made sure to block 3D, then 3D should be completely disabled, and the 3D light shouldn't be on.

semenogka, 2025-08-01:
> The 3D light is on, indicating that 3D is not really fully disabled. If Parental Controls is enabled and you specifically made sure to block 3D, then 3D should be completely disabled, and the 3D light shouldn't be on.

I tried to turn on parental control and block 3D, but it didn't help me